Infineon/DPS310-Pressure-Sensor #2: Wrong pressure value

devKlausS, 2017-08-04:
I am having problems reading out a correct pressure value using the i2c_command.ino example . The pressure value I receive is 86869888 Pascal. The Temperature value is correct ~27°C. Moreover the pressure value is increasing if I rise the sensor but the value should decrease.

mhollfelder, 2017-08-05:
Thank you very much for your feedback.

As we tested the library heavily with our XMC Arduino implementation and were not facing any issues, we assume that the problem has something to do with specific microcontrollers themselves.

Could you tell us which microcontroller you are using? Is it a 8-bit Arduino UNO?
We think it could be related to bit shifts which might be handled based on the toolchain and microcontroller differently, but we need to do further investigations based on your input.

Best regards,

Manuel

mhollfelder, 2017-08-08:
We added a fix by using a typecast (uint32_t) for the bit shift and now it is working on Arduino UNO as expected. A new version has been released.
Could you also test whether it is also working for your system correctly? Thank you very much!

Best regards,

Manuel

devKlausS, 2017-08-08:
I am using an Arduino Uno. The typecast will solve the problem. I fixed it the same way in fork today.

devKlausS, 2017-08-08:
Thx for the quick fix!

mhollfelder, 2017-08-09:
Great, thank you for the fast response.
Hopefully, this issue is solved by this fix also for other systems.

In case you face more problems, feel free to open a new issue ticket.

Best regards,

Manuel

closed 2017-08-09 by mhollfelder

kiminho9, 2021-01-26:
Hello, i have a question about dps310 high precision mode. I configured sensor in that mode, but i think that self-heating is problem. Values of pressure are continuously falling (little but falling). Did you test high precision mode ?

mhollfelder, 2021-01-26:
> Hello, i have a question about dps310 high precision mode. I configured sensor in that mode, but i think that self-heating is problem. Values of pressure are continuously falling (little but falling). Did you test high precision mode ?

Could you please kindly open a new issue and describe your problem there?
Moreover, could you also please provide an example reading?

Please provide:
- Hardware used as MCU
- Hardware used for DPS368
- Software used, especially example and library version

The mode was tested as well, but I will check how to help you here.

Thanks and best regards,

Manuel

JAVI2106, 2022-04-13:
Hola, tengo varios sensores DPS310 y DPS368, todos marcan mal la temperatura. He utilizado todos los programas que he encontrado y siempre todos dan el mismo error. Si partimos con 25º a los 0º marcan entre  -2.5º y -3º. Otros sensores como el BMP388, 280, TMP117, dan datos en el entorno de 0º.
He utilizado ESP-32 y ARDUINO UNO; el resultado el mismo.
¿ Alguien puede explicarlo ?.
Gracias, saludos.