Infineon/DPS310-Pressure-Sensor #4: This library doesn't work with Arduino Zero, MKR1000, NodeMCU

jayakarthigeyan, 2017-09-14:
This library works with Arduino UNO very well. But when we compile it for Arduino Zero or MKR1000 or NodeMCU it gives errors. Is there any solution to make this library work with Arduino Zero or MKR1000.

jayakarthigeyan, 2017-09-15:
no matching function for call to 'TwoWire::requestFrom(uint8_t&, unsigned int, uint8_t&, unsigned int, unsigned int)'

  if(m_i2cbus->requestFrom(m_slaveAddress, 1U, regAddress, 1U, 1U) > 0)

no matching function for call to uint8_t TwoWire::requestFrom(uint8_t, size_t, bool)

     uint8_t requestFrom(uint8_t address, size_t quantity, bool stopBit);

mhollfelder, 2017-10-07:
Sorry for the delayed reply, we have been working on the problem in the meantime as also other boards were affected.

We are preparing a new release (1.0.4) which was tested by me right now for the mentioned hardware:

- Arduino/Genuino Zero
- Arduino/Genuino MKR1000
- NodeMCU

One of the reasons was a non-consistent type of the function parameter variable types. 
We fixed this and hope that it will work for a broad range of available boards right now. 
However, I only could test the successful compilation, but no real-world use for the boards as the hardware is not available for me at the moment. 
Therefore, if you face additional issues, please let me know.

I will release the new version 1.0.4 as soon as possible for downloading in the Arduino IDE.
If you'd like to test it earlier, you could already download this repo and replace the files. 

Best regards,

Manuel

jayakarthigeyan, 2017-10-07:
Hi, 

I have a tested library based on Software I2C which I have tested in all the three boards, Arduino Zero, Arduino MKR1000 and NodeMCU with hardware. 

You can see this at the hackster project link given below:
https://www.hackster.io/jayakarthigeyan/smart-desk-clock-to-save-power-using-iot-at-home-office-cd1695

mhollfelder, 2017-12-12:
Thank you very much for your feedback and your work, looks really amazing!
In case you need help, feel free to open an issue here again.

Best regards,

Manuel

closed 2017-12-12 by mhollfelder