Infineon/DPS310-Pressure-Sensor #6: Initialisation failed on I2C esp32 wroom

Imara90, 2018-01-04:
Dear, 

I've set up the examples for I2C communication on the esp32 Dev Module board in the Arduino IDE. In all examples the object initialisation fails, or at least returns -2. I've programmed a loop to try all addresses, but to no avail. Any idea on what to do?

mhollfelder, 2018-01-08:
Hi,

Could you please help us and give some more information? 

- How do you wire the board?
- Which pins do you use for SDA and SCL?
- Which PCB do you use for the DPS310?
- Which example do you run to produce an error code?
- Are there errors during compilation or other warnings?

I will try to replicate the issue with an ESP32 board and try to help you with this issue.

Best regards,

Manuel

Imara90, 2018-01-09:
Hi Manuel, 

Off course. 

> How do you wire the board?

[image]

> Which pins do you use for SDA and SCL?
> Which PCB do you use for the DPS310?

The DPS310 is already on a pcb layout, so I'm just using the chip. It is connected as in the attached image. I'm using the esp32 wroom and pin IO22 and IO21 for respectively the I2C clock and data lines. The NFC reader from the images is currently not connected

> Which example do you run to produce an error code?

I've used both the I2C_background and the I2C_command 1.0.5 example from the arduino library manager. I've selected the ESP32 dev module as the board to develop for. 

> Are there errors during compilation or other warnings?

Yes, 

Arduino\libraries\DPS310\src\Dps310.cpp:1408:12: error: 'MSBFIRST' was not declared in this scope

In order to fix this, I had corrected the MSBFIRST to the SPI_MSBFIRST version of the esp32 library. I thought it however shouldn't matter at all since I'm not using the SPI part of the library.

Thanks for having a look!

mhollfelder, 2018-01-09:
Thank you very much.
I will have a look and come back to you.

Best regards,

Manuel

mhollfelder, 2018-01-16:
I have now hands on an ESP32 evaluation board and will check the issue.
As soon as I have an update, I will let you know.

Best regards,

Manuel

Imara90, 2018-01-16:
Perfect, thanks a lot

Imara90, 2018-01-19:
Hey Manuel, 

Got back into trying. It seems that the error occurs on initialisation of the DPS310. When trying to read either temperature or pressure, I get a return value of -2. Any luck on your side?

Best,
Imara

Imara90, 2018-01-23:
Hey Manuel, quick update. Seems that the factory misinterpreted the vent hole as an indicator for the first pin. I've desoldered them and the I2C lines look ok again. Tomorrow I'll receive new DPS310's and check whether the MSBFIRST to SPI_MSBFIRST tricks did actually work and I'll get back to you. Sorry for any incovenience!

mhollfelder, 2018-01-23:
We had troubles with the ESP32 board and needed to fix it before we could start testing with the DPS310.

This means that your boards with the DPS310 are not soldered properly? 
It would also explain the error code -2, i.e. that there are troubles with reading out the sensor.
In the meantime, I will try to test it with the ESP32 to ensure that there are no other troubles.

Thanks a lot for this information with the vent hole, I will forward it to the responsible persons internally as also other ones might have this issue.

Best regards,

Manuel

Imara90, 2018-01-24:
Hey Manual, 

I soldered another DPS310 on and although I'm finally seeing a compatible data and clock line on my logic analyzer, I'm getting only NACKs and again a return value -2 indicating that initialisation did not work. Any luck on reproducing this?

mhollfelder, 2018-01-24:
Hey, 

yes, it is working here, we got it running right now.

Setup:
- [ESP32 Dev Board](https://github.com/espressif/arduino-esp32#esp32dev-board-pinmap) Compatible evaluation board
- Current ESP32 Arduino core
- Needed to add `#define LSBFIRST 0` and `#define MSBFIRST 1` to DPS310 library `dps310_consts.h` as this was not declared in the ESP32 SPI library
- Connected our [Shield2Go](https://www.infineon.com/cms/de/product/evaluation-boards/s2go-pressure-dps310/) according to the pin out of the ESP32 Dev Board: 3.3V, GND, SDA and SCL
- Flashed the `i2c_command.ino` example

Could read out the pressure and the temperature without any problem. 
Can you read out the DPS310 if you wire the board with an Arduino Uno? 
Just to check that the pressure sensor is working at all. 

Have you tried applying a PWM with 500 kHz and watched the signal quality or do you have a scope capable of recording the signal when trying to communicate?

We will get it working together :)
Lets try to find out whats going wrong.

Best regards,

Manuel

Imara90, 2018-01-24:
Hey Manuel, 

Thanks for your reply. Good to know that you've got it working! Still no cigar on this side though. I've adjusted the library and flashed `i2c_command.ino` but it returns -2. I've also flashed an i2c scanner as seen here [https://playground.arduino.cc/Main/I2cScanner](url), which does return the address of my nfc reader, but not the barometer (both when plugged in singularly). The i2c signal looks good, I've added a printscreen of my logic analyzer. 

[image]

Any other ideas?
thanks!
- Imara

mhollfelder, 2018-01-24:
Hey Imara,

some more ideas/questions:

- You don't have a scope to not only see the logic, but also the quality of the signal, i.e. rise time, etc.?
- Just a crazy idea, have you tried using the alternative address 0x76?
Maybe the floating SDO pin causes some disturbance?
- You tested multiple resoldered DPS310 now, right?
- What value has VCC?

Thanks a lot and best regards,

Manuel

Imara90, 2018-01-24:
Yes, thanks!

I do have a scope, quality of the signal also looks good. I attached some images, excuse the skewness of them :)

[image]
[image]

The i2c scanner program scans all possible 8-bit addresses, so both 0x76 and 0x77. 
I've tested 2 resoldered DPS320s, I don't have any mroe available right now. 
VCC is 3.27V

mhollfelder, 2018-01-24:
Hi Imara,

I think there is a problem with the soldered DPS310 as all the rest seems okay.
My assumption is that the sensor is not soldered properly, broken or something else is going wrong.

How can I help you?
Should I try to trigger an evaluation board for you so that you test the system-setup?

Thank you very much for the information and the pictures!

Best regards,

Manuel

Imara90, 2018-01-25:
Hey Manuel, 

I'll first try and desolder and solder them again before we need an evaluation board. Any recommendation on the heatgun temp? After too long, the casing of the DPS310 starts to melt at 290-300 degrees. 

Best, 
Imara

Imara90, 2018-01-25:
Oh God, finally! Resoldered one of them for the fourth time and it's getting valid results. 

Thanks so much for the help. I think the real issue here was the factory misinterpreting the vent hole for first pin. 

Best regards,
Imara

closed 2018-01-25 by Imara90

mhollfelder, 2018-01-25:
Perfect, glad to hear that!
If you need any further help, please let me know.

Best regards,

Manuel