Infineon/DPS310-Pressure-Sensor #15: correctTemp - bibliography

jeanguerin, 2019-03-19:
Hello,

I have unknown issue with DPS310 integration in a watch. In some watches, not all of them, we observe 2 incoherent patterns which seem to come from the pressure sensor. 
We are working at 1Hz with an oversampling of 16.

1. around important variation, some absurd dots appears (for a variation of 1 meter (8 Pa), 1 measure go up to to 25 Pa). 
2. after the variation, it could take 3-7 seconds for the altimeter to reach the stabilized value

 
[image]

I am now stuck with the issue and I am considering a bad calibration of temperature and the usage of the correctTemp function you offer. 
As the function mainly write into registers which are not define in datasheet, is there a paper confirming the usage of this command?

Thank you for your support,

JG

mhollfelder, 2019-03-21:
Hi JG,

Thank you very much for your report.
Could you give some more details about the report, please?

1. Alti is a pressure or temperature value in a.u. or already processed? 
Please, more details about the graph.
2. Are you completely using the Arduino library in your approach and only this library is used for DPS310?
3. Do you read out the temperature every time before reading out the pressure value?
4. Do you have custom boards and did you use vapor phase soldering?
5. Is the device completely without enclosure and the DPS310 is exposed?
6. Is this reproducible with the device, e.g. every time the same behavior is seen?
7. Do you see some drifting of the effect or time dependency? 

Especially, 3. is quite important for further discussion, but answers to all the questions help me to proceed.

The correctTemp() function does not harm the sensor in any case - even if the sensor does not have the problem. It would just reset these registers and not change the expected behavior.

I think we need to do some investigation and once I get your reply, I am happy to help you out.

Thanks a lot and best regards,

Manuel

jeanguerin, 2019-03-23:
Hello,

Thank you for your answer.
1. Alti (for Altimeter), is the pressure in Pa.
2. In fact, we are not using directly the Arduino library, we rewrote for MCU EFM32, and we are only using altimeter functions
3. We read temperature every time before calculation of pressure
4. We are working on our own custom board, communicating with I2C, which are soldered in an oven. As I have many default on current production, I would say this could be an interesting lead.
5. The board is in the watch with a membrane. We have done the testing with close watch then with watch open and saw no difference.
6. We observed with this device that after a certain amount of reset (not at first one), the problem disappears. But our feeling is that it comes back more or less randomly. All this is fuzzy, we have very few real observation of it.
7. Can't say. 

About correctTemp(), do I do it once in the lifetime of the DPS310, or at each reset to be sure?

At your disposal if you need me to do further testing.

JG

amigueti, 2019-05-28:
Hello.

I´m having some similar issue. I recently bought a 'Pressure 3 click' board which includes a DSP310 sensor. I wrote my own library for a MCU MSP432. 

I´ve started doing some tests and got problems with the temperature measures. I´m always measuring 39.xy ºC no matter the conditions. I copied your 'correctTemp()' solution and then restarted my tests. The thing is than when I use your solutions I always measure temperatures between [-3, 2] ºC. 

The question is, could the issue be in my code? In the board? Or maybe, in the sensor? Tomorrow I´ll try the same board with an Arduino and your library. And write back my results.

boramonideep, 2023-06-19:
The DPS310 and DPS368 repositories are being consolidated into a single repository here: https://github.com/Infineon/arduino-xensiv-dps3xx

Please open an issue there if the problem still persists.

closed 2023-06-19 by boramonideep