Infineon/DPS310-Pressure-Sensor #16: Unstable pressure value

ClementBC, 2019-07-19:
Hello,

I have an issue with the pressure sensor, as you can see on the picture below, the pressure measures are not correct. (At ~20sec the sensor move from a position A to a position B (position A higher than B) so I should have stable measures before ~20sec, a gap and stable measures with higher values after ~20sec). 

[image]

The pressure sensor configuration is :
- for the pressure 4Hz with an oversampling of 2
- for the temperature 4Hz with an oversampling of 1
- interrupt when the FIFO is full, we read the data in the FIFO at every interrupt.

The sensor is on a custom board (with an MCU EFM32). I have several custom boards with this sensor this problem appears only on a few boards. In the defective board I have pressure value at 97600-97900 but with a good board I have value at ~101400.

The communication between the MCU and the sensor is correct, I can read or write into the registers.

Did you ever see a problem like this one and what can be the cause ?

Thank you.
ClémentBC

pandayswarnam, 2019-07-19:
Hi @ClementBC ,

Kindly refer to the issue in this [Link](https://github.com/Infineon/DPS310-Pressure-Sensor/issues/15 ). I think there is some problem with EFM32 MCU. We will get one and test on that and then come back to you.

Thank you,
Swarnam

ClementBC, 2019-07-19:
Hi Swarnam,

Thank you for your quick report 🥇.

I checked up the sensor under a microscope and the sensor has no cover cap, so I can see “inside” the DPS310 and one of the connection is out.
[image]
_(sorry not great picture)_

I think the pressure value return by the sensor is weird because of this connection. I can still interact with the sensor despite this fault connection consequently can I know if there is a fault connection directly in my code ?

Thanks
CBC

closed 2019-09-16 by pandayswarnam