Infineon/DPS310-Pressure-Sensor #23: Sample values for DPS310 coefficient, temperature registers

ojousima, 2020-11-25:
Hello,

I'm writing unit tests for DPS310, and as a part of tests I'm mocking the register map to check the coefficient calculation and further temperature + pressure calculations. 

Could you provide a sample data set with coefficients, positive temperature value, negative temperature value and a temperature compensated pressure readout? I'd need the raw register values and parsed values.

Thanks

ojousima, 2021-02-16:
[image]

tmanaud, 2022-11-03:
Here you are, coefs from my device, with Pcom in float and int value, you can copy paste in matlab/octave:
```
c00= 80074
c10= -56035
c20= -9919
c30= -1321
c01= -3145
c11= 1432
c21= 152
c0= 223
c1= -291
kP= 3670016
kT= 3670016

Praw= -1509278
Traw= 156656

Traw_sc = (Traw /kT);Praw_sc = (Praw /kP);Pcomp = c00 + Praw_sc*(c10 + Praw_sc*(c20+ Praw_sc*c30)) + Traw_sc *c01 + Traw_sc *Praw_sc*(c11+Praw_sc*c21)
Pcomp_float = (((c00)/1 + (Praw*c10)/(kP) + (Praw*Praw*c20)/(kP*kP) + (Praw*Praw*Praw*c30)/(kP*kP*kP) + (Traw*c01)/(kT) + (Traw*Praw*c11)/(kP*kT) + (Traw*Praw*Praw*c21)/(kP*kP*kT)))
Pcomp_int = (int64)((c00/1) + (int64)((Praw*c10)/(kP)) + (int64)((Praw*Praw*c20)/(kP*kP)) + (int64)((Praw*Praw*Praw*c30)/(kP*kP*kP)) + (int64)((Traw*c01)/(kT)) + (int64)((Traw*Praw*c11)/(kP*kT)) + (int64)((Traw*Praw*Praw*c21)/(kP*kP*kT)))
```

ojousima, 2022-11-03:
Thanks :) 

I wrote a C-language driver for embedded systems, it needs only 3 functions from user: sleep, write and read. 
The driver I wrote is available [here](https://github.com/ruuvi/ruuvi.dps310.c) under MIT license in case someone finds it useful. 

I ended up using guessed coefficients and matched the temperature values on those, since the driver fills my needs I probably won't be updating the test coefficients. 

Closing this issue for now.

closed 2022-11-03 by ojousima