Infineon/DPS310-Pressure-Sensor #24: dps310 pressure sensor high precision mode (5cm)

kiminho9, 2021-01-27:
Hello i have a question about dps310 high precision mode. I configured sensor in that mode, but values i got are oscillate to much. I bought pressure 3 click board with dps310 sensor, and i am using Arduino libraries from this site. For example , my first 10 values are 985.31mBar , then values are start rising slowly ( i get values on every 250ms) 985.32 , 955.33 ... 985.35 , then again falling 985.34, 985,33 ... 985.31 ... 985.28mBar ( values are oscillate about +-50cm in relation to the starting values).  Why ? Is it problem in self heating ? Why values oscillate when conditions are same ? Where did you test high precision mode ? In closed room or outside ? Did someone have similar problem ? Do i need some extra functions to eliminate this problem ? Regards :D

mhollfelder, 2021-01-27:
Could you please share the code or example which you are using?
This would already help a lot and also help to sort out any problems.

If you read only the temperature once (or never) in your code, the last internal temperature value will be used for compensation.
Hence, the driver expects also to read out the temperature to update the value inside the driver (SW side).

More details here from the driver::

https://github.com/Infineon/DPS310-Pressure-Sensor/blob/888200c7efd8edb19ce69a2144e28ba31cdad449/src/Dps310.cpp#L75 

Hence, sharing the used example and/or software would help to trace the issue and propose a solution.

Thanks and best regards,

Manuel

kiminho9, 2021-01-27:
This is my code, it is the same like in your library except configuration of sensor, and i divide values of pressure in Pascal with 100 to get values in mBar, and i put delay 250ms on the end instead of 10000 because i want to see values on every 250ms . Thank you for help.

`#include <Dps310.h>

// Dps310 Opject
Dps310 Dps310PressureSensor = Dps310();

void setup()
{
  Serial.begin(9600);
  
  while (!Serial);

  //Call begin to initialize Dps310PressureSensor
  //The parameter 0x76 is the bus address. The default address is 0x77 and does not need to be given.
  Dps310PressureSensor.begin(Wire, 0x76);
  //Use the commented line below instead to use the default I2C address.
  //Dps310PressureSensor.begin(Wire);

  //temperature measure rate (value from 0 to 7)
  //2^temp_mr temperature measurement results per second
  int16_t temp_mr = 2;
  //temperature oversampling rate (value from 0 to 7)
  //2^temp_osr internal temperature measurements per result
  //A higher value increases precision
  int16_t temp_osr = 2;
  //pressure measure rate (value from 0 to 7)
  //2^prs_mr pressure measurement results per second
  int16_t prs_mr = 2;
  //pressure oversampling rate (value from 0 to 7)
  //2^prs_osr internal pressure measurements per result
  //A higher value increases precision
  int16_t prs_osr = 6;
  //startMeasureBothCont enables background mode
  //temperature and pressure ar measured automatically
  //High precision and hgh measure rates at the same time are not available.
  //Consult Datasheet (or trial and error) for more information
  int16_t ret = Dps310PressureSensor.startMeasureBothCont(temp_mr, temp_osr, prs_mr, prs_osr);
  //Use one of the commented lines below instead to measure only temperature or pressure
  //int16_t ret = Dps310PressureSensor.startMeasureTempCont(temp_mr, temp_osr);
  //int16_t ret = Dps310PressureSensor.startMeasurePressureCont(prs_mr, prs_osr);

  if (ret != 0)
  {
    Serial.print("Init FAILED! ret = ");
    Serial.println(ret);
  }
  else
  {
    Serial.println("Init complete!");
  }
}

void loop()
{
  uint8_t pressureCount = 20;
  float pressure[pressureCount];
  uint8_t temperatureCount = 20;
 float temperature[temperatureCount];

  //This function writes the results of continuous measurements to the arrays given as parameters
  //The parameters temperatureCount and pressureCount should hold the sizes of the arrays temperature and pressure when the function is called
  //After the end of the function, temperatureCount and pressureCount hold the numbers of values written to the arrays
  //Note: The Dps310 cannot save more than 32 results. When its result buffer is full, it won't save any new measurement results
  int16_t ret = Dps310PressureSensor.getContResults(temperature, temperatureCount, pressure, pressureCount);

  if (ret != 0)
  {
    Serial.println();
    Serial.println();
    Serial.print("FAIL! ret = ");
    Serial.println(ret);
  }
  else
  {
   // Serial.println();
    //Serial.println();
   // Serial.print(temperatureCount);
    //Serial.println(" temperature values found: ");
    for (int16_t i = 0; i < temperatureCount; i++)
    {
      Serial.print(temperature[i]);
     // Serial.println(" degrees of Celsius");
    }

   Serial.print("----");
   
    //Serial.print(pressureCount);
  //  Serial.println(" pressure values found: ");
    for (int16_t i = 0; i < pressureCount; i++)
    {
      Serial.print(pressure[i]/100);
      
    }
     Serial.println();
  }

  //Wait some time, so that the Dps310 can refill its buffer
  delay(250);
}`

mhollfelder, 2021-01-27:
I edited your comment to add proper code formatting with ` ``` ` .

Code looks good and it updates the internal temperature value as well based on my first findings.
So, temperature compensation should be working in principal.

Does this behavior stay constant with oscillation independent of runtime or get better after some time?
As the pressure sensor is very sensitive, could you try to shield it somehow, e.g. with some box or stiff bag around it, and check afterwards whether it changes its behavior or gets better?

We saw in other applications that due to its high sensitivity it catches up tiniest pressure changes, e.g. it could be some ventilation or pressure changes inside the room with oscillations.

If the problem was related to self-heating, one would expect an asymptotic behavior for `t -> ∞` with constant readings and setup (or close to ∞, e.g. 5 minutes :D ), e.g. that it has a fixed offset from the initial values at some time.

Best regards,

Manuel

kiminho9, 2021-01-27:
O.k. , i  will try to close sensor in some box, then test it for longer time and see what happening. I will tell you results after that. Thanks for help! Regards :D

mhollfelder, 2021-01-27:
Perfect :)

Afterwards we can jointly analyze and check the results.
It would also be extremely helpful if you could collect the series via `Serial.print()` with a time-stamp, e.g. via `millis()`.
If you share the files with me, I can do some further checks of the values and have a look.

Thanks and best regards,

Manuel

kiminho9, 2021-01-28:
Hello. I tested sensor in closed box, in closed room without air-condition and other heating devices and i got almost perfect results(5cm the most of the time, sometimes +-(10-15 cm) but then return to 5cm). I think the main problem is that the sensor is too sensitive. I saw video on YouTube [](https://www.youtube.com/watch?v=Bmiox3HII6g) and i don't know how they got that stability of measurements. Maybe i need some extra software tool to reduce that sensitivity? Thanks, best regards :D

Telepresent, 2021-09-02:
Could you provide the code to calculate height variation based on pressure and temperature (if needed)? Thanks!

mintisan, 2022-09-13:
Why 5cm?
`• Pressure sensor precision: ± 0.002 hPa (or ±0.02 m) (high precision mode).`From datasheet 2020-10-15 version.

closed 2023-04-05 by jaenrig-ifx