Infineon/DPS310-Pressure-Sensor #27: Errors in temperature readings in 1/200 devices

gjpartis, 2022-01-25:
**Prerequisites**

On particular devices, the temperature reading can be around 30-40degC higher than expected.

**Description**:

This is a known defect. There is however no errata sheet. The is no "paper trail" for audits.

The "recommended" fix does not help in mass production of products.

_"There could be a problem with temperature measurement of the DPS310. If your DPS310 indicates a temperature around 60 °C although you expect around room temperature, e.g. 20 °C, please call the function correctTemp() as included in the library to fix this issue."_

Poking values into random (ie, undocumented, magic) registers on devices which demonstrate this fault may be okay for hobbyists, but is not really a solution.

Can a formal explanation be supplied (your erratum@infineon.com dont reply to formal requests) with a documented automated process to _**determine when**_ these magic numbers are to be used (checking internal batch or revision number et al)?

This fault appears in around 1 in 200 sample devices.

mhollfelder, 2022-02-01:
Dear gjpartis,

Thank you very much for providing the above details about your issue.
First of all, sorry to hear that you are facing such problems and haven't received the expected information so far.

We would need additional information about the batch of the products, how the pressure sensors are assembled, etc.
With these details, we will be able to provide you feedback and documentation on how to proceed.

In order to guide you to the right internal contact for a direct exchange, please open a request at the Technical Assistance Center (TAC) here:

https://www.infineon.com/tac

You can also refer in your description to me directly (Manuel Hollfelder) so that I am getting directly involved in the process.
Please let me know once you have opened a ticket so that I can trigger additional steps.

Thanks and best regards,

Manuel

gjpartis, 2022-02-04:
Hi Manuel

I have raised a ticket at Infineon's TAC portal, and mentioned you by name.

We have now found an "off the shelf" DPS310 dev board from Mikro Elektronika in one of our Indian offices which also demonstrates this fault. This means our prototype manufacturing process (an external company in Lisburn, Northern Ireland) is unlikely to be the problem.

Kind regards

Gary Partis

gjpartis, 2022-02-14:
Hi Manuel

Is there any traction on the ticket I raised via the TAC portal, pertaining to the DPS310 erroneous temperature readings?

Kind regards

Gary Partis

gjpartis, 2022-03-10:
Hi again Manuel

Is it worthwhile to simply not use the DPS310 in any of our designs; as it appears to be a faulty design which can not be fixed via software automation; having to rely on manual checking and then applying some fix?

Kind regards

Gary Partis

mhollfelder, 2022-03-10:
Hi Gary,

Please let me quickly do an internal follow-up.
I provided internally feedback, but please let me check the status.

In due time, I will come back to you.

Thanks and best regards,

Manuel

closed 2023-06-19 by boramonideep